ThunderFly-aerospace/TF-G1 #271: Maximalizace efektivní plochy SOP

kaklik, 2019-11-19:
Boční průmět směrovky je potřeba maximalizovat tak, aby směrovka měla největší možnou účinnost. 
Limitem jsou rozsahy rotoru a záklon a stlačení podvozku. 

[image]

Je několik možností, jak plochu zvětšit 
- Změna úhlu mezi polovinami ocasních ploch
- Přidání nástavců na horní případně spodní část 
- Změna tvaru ocasních ploch 

### Změna úhlu mezi směrovkami 

Aktuální úhel mezi oběma polovinami směrovek je 90 stupňů jeho snížením se zvedne boční průmět směrovky. Můžeme vyzkoušet například 60 stupňů. 

### Nástavce na horní část 

Na horní část směrovky by bylo možné přidat šikmý nástavec, který by zvětšoval efektivní plochu směrovky, ale nekolidoval by s rotorem. 

[image]

### Změna tvaru profilu ocasní plochy

Profil ocasních ploch nemusí být všude stejně tlustý. Proto by bylo možné v místech, kde tlustý musí být (v oblasti serva) jeho tloušťku upravit tak, aby pohyblivé části bylo možné prodloužit, bez ztráty aerodynamičnosti profilu.

V extrémním případě by tak profil jedné poloviny směrovky mohl vypadat nějak takto: 

[image]

Pro změnu tvaru lze pravděpodobně využít funkce z [této knihovny](https://www.thingiverse.com/thing:900137).

kaklik, 2019-12-26:
Mezi zdrojové soubory jsem přidal knihovnu sweep a example, který v ní je upravil tak, aby více odpovídal našemu použití.
Pro vytvoření zakřivené plochy jsem jako parametr použil [gaussovu funkci](https://en.wikipedia.org/wiki/Gaussian_function). 

[image]

kaklik, 2019-12-26:
Do knihovny jsem přidal i příklad pro válec s proměnlivým průměrem. 

[image]

kaklik, 2020-01-23:
Informace od @taury55  ke commitům výše:
Je tam problém s menším modelem té celé směrovky (666_1028_rudder_shape_inside). Je stejně široký. Nepřišel jsem na to jak ho zúžit o šířku stěny směrovky (Rudder_wall_thickness)

kaklik, 2020-02-19:
U modelu zřejmě nějak nefunguje propagování proměnné draft a ovlivnění kvality renderování. 
Neboť i tiskové modely jsou stále hranaté. 

[image]

@taury55  model zřejmě ještě nijak neřeší vnitřní strukturu (žebra a výplně), je to tak?

taury55, 2020-02-19:
Známé chyby na modelu:

- [ ] Nějak jsem přehlédl, že draft nefunguje. Kouknu se na to, ale bude to chyba v části Naca kódu, takže možná budu potřebovat pomoc. To mě přivádí k druhé chybě.
- [ ] Nejsem si jist, jestli funguje kód pro generaci směrovky správně. Díky mým parametrům. Tam bych potřeboval poradit. Konkrétně ohledně zmenšení směrovky o šířku stěny.

```
module 666_1028_rudder_shape()
{
  N = 40;
  sweep(gen_dat(M=tail_height / 2, dz=1,N=N), showslices = false);

  // specific generator function
  function gen_dat(M=10,dz=0.1,N=10) = [for (i=[0:dz:M])
    let( L = extra_length(i))
    let( af = vec3D(
        airfoil_data([0,0,0.05+thickness(i)], L=tail_length+extra_length(i), N = N)))
    T_(0, 0, (i)*2, af)];  // translate airfoil

  function thickness(i) = gauss(i,0.05,(tail_bottom_height + Rudder_height / 2) / 2,15);
  function extra_length(i) = gauss(i,50,(tail_bottom_height + Rudder_height / 2) / 2,23);
}

module 666_1028_rudder_shape_inside()
{
  //TODO: wall thickness repair
  N = 40;
  sweep(gen_dat(M=tail_height / 2 - Rudder_wall_thickness / 2, dz=1,N=N), showslices = false);

  // specific generator function
  function gen_dat(M=10,dz=0.1,N=10) = [for (i=[0:dz:M])
    let( L = extra_length(i))
    let( af = vec3D(
        airfoil_data([0,0,0.05+thickness(i)], L=tail_length+extra_length(i)-Rudder_wall_thickness*2, N = N)))
    T_(Rudder_wall_thickness, 0, (i)*2 + Rudder_wall_thickness, af)];  // translate airfoil

  function thickness(i) = gauss(i,0.05,(tail_bottom_height + Rudder_height / 2 - Rudder_wall_thickness / 2) / 2,15);
  function extra_length(i) = gauss(i,50,(tail_bottom_height + Rudder_height / 2 - Rudder_wall_thickness / 2) / 2,23);
}
```

- [ ] Válec náběhu směrovky je užsí než by měl být. V návaznosti na 2. problém.

```
//Rudder_shaft_outside_shape--------------------------------

module 666_1028_rudder_shaft_shape(){
  N = 40;
  sweep(gen_dat(M=tail_height / 2, dz=1,N=N), showslices = false);

  // specific generator function
  function gen_dat(M=10,dz=0.1,N=10) = [for (i=[0:dz:M])
    let( af = vec3D(circle_data(r=Rudder_shaft_outside_min_diameter/2+thickness(i), N=50)))
    T_(0, 0, (i)*2, af)];

  function thickness(i) = gauss(i,(Rudder_shaft_outside_max_diameter-Rudder_shaft_outside_min_diameter),(tail_bottom_height+Rudder_height/2)/2,15);
}

//Rudder_shaft_inside_shape---------------------------------

module 666_1028_rudder_shaft_shape_inside(){
  N = 40;
  sweep(gen_dat(M=tail_height/2-Rudder_wall_thickness/2, dz=1,N=N), showslices = false);

  // specific generator function
  function gen_dat(M=10,dz=0.1,N=10) = [for (i=[0:dz:M])
    let( af = vec3D(circle_data(r=Rudder_shaft_outside_min_diameter/2+thickness(i)-Rudder_wall_thickness, N=50)))
    T_(0, 0, (i)*2+Rudder_wall_thickness, af)];

  function thickness(i) = gauss(i,(Rudder_shaft_outside_max_diameter-Rudder_shaft_outside_min_diameter),(tail_bottom_height+Rudder_height/2)/2,15);
}
```

- [ ] Absence díry na zansunutí osy směrovky. Zapomněl jsem na ni. Ale není náročné ho opravit.
- [ ] Žebra jsem odebral, protože jsem měl za to, že se přidají v přípravě na tisk (Nevím, jak proces funguje.), ale nemělo by být težké je přidat. Výplně (V modelu dutiny) jsou v modelu zahrnuty. To byla ta nejtežší část. Jen jsem nevymodeloval dutinu do nástavce na tyč (viz. obrázek výše), protože jsem měl za to, že se výplň přidá v postprocesingu, jelikož nástavec jsem zamýšlel celý vyplněný.

Takže problém čtvrtý vyřeším, první problém se pokusím vyřešit a pátý můžu bez problém opravit, jestli je potřeba. Ale s druhým a třetím bych potřeboval pomoci s Naca knihovnou.