ThunderFly-aerospace/TF-G1 #279: Úprava generování STL a AMF

roman-dvorak, 2020-01-12:
Dosavadní makefile generoval pouze stl soubory. Tyto soubory se pak využívaly k tisku a využívaly se v sestavách.

U některých modelů (např tištěný rotor) by bylo vhodné mít možnost používat modifikátory k automatizování používání modifikatorů je potřeba právě AMF (více o tom v #278)

Dále by to chtělo prozkoumat, jestli je potřeba generovat i STL. Nebo jestli dokážeme STL plně nahradit s AMF. Popřípadě jak by taková konverze měla vypadat.

kaklik, 2020-01-12:
@JanKott  Podíval by ses na to?

JanKott, 2020-01-20:
Nějak nerozumim o co jde. Můžete mi to trochu rozvést?

roman-dvorak, 2020-01-20:
Jo, v tomto issue není přesně popsáno, co by bylo potřeba.

K slicování dílů používáme Slic3r. Ten byl vybrán z důvodu, že jako jediný umí nějakým způsobem pracovat s konfiguračními soubory a z přikazové řádky. 

Při slicování s modelem udělá dvě zásadní věci a to to, že ho přesune v ose Z tak, aby přesně seděl na podložku a pak model vycetruje v X, Y. Tyto translace jsou neznámé.

Bylo by ideální aby se tyto informace ukládaly přimo do gcodu. Znalost této translace je potřebná třeba při výpočtu těžiště modelu z gcodu. Pokud je infill velmi řídký, tak se těžiště plného tělesa (STL/AMF) může o dost lišit oproti těžišti gcodu. 

Dokud jsme používaly STL, tak jsem udělal skript, který si určil hranice STL soubory. Následně to porovnal s gcodem a podle toho odhadl všechny tři posuvy. Ale s použitím AMF to zatím neumím a je to strašně nesystémové řešení. Proto by to bylo nejlepší upravit v Slic3eru.

JanKott, 2020-01-22:
Zkusim se kouknout do zdrojáků, ale jen tak cvičně - tohle by nestačilo:

--export-gcode will center the model around the centroid point of the configured print bed (**unless the --dont-arrange option is supplied: original coordinates will be used in that case**) ?

roman-dvorak, 2020-01-22:
Ano, zarovnání okolo středu - tak to je výchozí nastavení. A tam právě vznikají ty tři neznámé transformace. 

Vypnutí posuvu je také možné. Tam je však velmi slžité zajistit v modelech, že jsou umístěné nějakých 7.5 a 7.5cm mimo počátek souřadnic. A 0mm v ose Z. Naopak většina modelů se modeluje v počatku souřadnic (není to pravidlo).

Na základě těchto předpokladů právě funguje ten můj skript pro STL, který tu transformaci odhadoval. Pak je další věc, když se bude tisknout na více různých tiskárnách. Jedna má podložku 15x15, další 18x18 a pak 20x20 či více. A bylo by fajn, aby to bylo vždy ve středu. (tohle je ostatně také trochu nedořešené z pohledu generování gcodu, kde se využívá pouze jeden konfigurák pro všechny.

kaklik, 2020-01-22:
Těch konfiguráků pro tisk si dovedu představit, že bude více. Každopádně je ale nežádoucí, aby se při tisku na jiné tiskárně muselo šahat do modelu.

kaklik, 2020-02-05:
Tady je testovací příklad (AMF a STL), který vygeneruje kostku spodní stranou 100mm nad podložkou v počátku souřadnic. 
Další alternativní posuvy je třeba dodatečně vyzkoušet: 

[test.zip](https://github.com/ThunderFly-aerospace/TF-G1/files/4160663/test.zip)

[image]

JanKott, 2020-02-13:
Doplnil jsem zapicosvání posunu a rotace do gcode souboru. Na příkladu to funguje, až to zkusíte na něčem složitějším, dejte vědět.

closed 2020-02-13 by JanKott