ThunderFly-aerospace/TF-G1 #282: Rozdělení horizontální osy mezi písty předního kolečka

kaklik, 2020-01-23:
V současné konstrukci je horní uchycení pístů realizováno relativně dlouhou závitovou tyčí, která je horizontálně prostrčena ložisky v prostředním otočném dílu. Na tuto závitovou tyč jsou z obou stran nasazeny držáky svislých pneumatických tlumičů.

Tato konstrukce je nevýhodná, neboť vyžaduje při dotahování/povolování použití nejméně dvou plochých klíčů zároveň (aby se zabránilo protáčení matice na druhé straně).

Vhodnější by bylo ložiska přesunout do držáků válců, případně kuličková ložiska úplně zrušit a nahradit je pouze kluznou vložkou (klidně vytištěnou plastovou), která bude přitažena šroubem k prostřednímu otočnému dílu kolečka. 

Pro uchycení pneumatických válců můžou být klidně použity šrouby průměru M4  (pro každý pneumatický válec jeden) zašroubované do matek v kapsách otočného dílu.

roman-dvorak, 2020-02-04:
Nákres řešení rozdělení horní horizontální osy předního podvozku z použitím kluzné vložky.
 
[image]

cesilko1, 2020-02-04:
šel by tam použít i lícovaný šroub?

kaklik, 2020-02-04:
Lícovaný šroub nemáme takhle krátký, museli bychom ho koupit. Proč? Zdá se ti to v něčem lepší?

cesilko1, 2020-02-04:
lícovaný šroub by lépe rozkládal sílu na díl připevněný k pístu

kaklik, 2020-02-04:
> lícovaný šroub by lépe rozkládal sílu na díl připevněný k pístu

Jak to? Lícovaný šroub o průměru 6mm bude rozkládat sílu úplně stejně jako mosazná vložka s průměrem 6mm, ne?

roman-dvorak, 2020-02-04:
Rozklad sil na díl připevněný k pístu by to právě moc velký rozdíl být neměl. Protože pouzdro má skoro stejný vnější průměr.

Navíc tady, když se z obou stran pouzdra dají podložky, tak ten díl nad pístem se bude moct na ose volně otáčet. Což je výhoda. 

A jestli se Ti to zdá málo robustní.. :) Tak na druhé straně (u igus ložiska) je to připevněné pomocí M3 šroubu.

roman-dvorak, 2020-02-05:
@cesilko1 vidím nové commity? Je to už podle Tebe nějak vyřešené? (jako že si to mám prohlédlout :) ) Nebo na tom ještě plánuješ pracovat?

cesilko1, 2020-02-05:
Mrkni na to ;)

kaklik, 2020-02-13:
Nově upravený díl při tisku neleží na podložce. 

[image]

cesilko1, 2020-02-16:
v [888_2024.scad](https://github.com/ThunderFly-aerospace/TF-G1/blob/improved_suspension/src/888_2024.scad) stačilo na konci přidat parametr `888_2024(1);`

kaklik, 2020-02-17:
Dobře, ale proč to není, nebo nefunguje v tiskové  sestavě? 
https://github.com/ThunderFly-aerospace/TF-G1/blob/improved_suspension/src/4printing/print_plate_888_2024.scad

Ten gkód ze kterého je generován obrázek výše vzniknul standardním procesem  (spuštěním make)  A jeho výsledek není dobrý, což znamená, že někde pocestě něco nefuguje..

roman-dvorak, 2020-02-17:
Já to tam teď vidím v pořádku.

cesilko1, 2020-02-27:
už jsem posunul ty písty

roman-dvorak, 2020-03-02:
Nové kolečko vypadá fakt dobře! Ještě je tam trochu problém s délkou dílů. Více v #269

commit 6073c11b (2020-02-28): Increase spacing between moving parts.

FIX #282
closed 2020-03-02 by roman-dvorak