ThunderFly-aerospace/TF-G1 #283: Workflow pro generování výrobních dat

roman-dvorak, 2020-02-07:
Protože se projekt TF-G1 velmi rozrůstá a vzniká mnoho podpůrných projektů, které jsou založeny na 3D tisku, začíná se stávat, že aktuální postup přípravy pomocí makefilu je nedostatečný a občas velmi složitý. Je potřeba připravit systém, který ze zdrojových dat (dneska primárně openscad). Tyto zdrojové formáty se můžou rozrůstat a mírně modifikovat. Obdobný návrh vzešel ze schůze s @kaklik  a @ChroustJan.

Celý systém musí umět několik základních věcí:

 * Načítat zdrojová data. V lepším případě umět nějaké mírné modifikace
 * Získání parametrů (jejich číselných výsledků) z OpenSCADu a parametrických souborů
 * Renderovat zdrojové soubory do formátu založených na mřížích. (Ideálně AMF, STL)
 * případnou konverzi formátů
 * Slicování modelů s nastavením sliceru specificky pro model
 * Vytváření modifikátorů pro slicery. Příprava pro jejich následovné použití vícekrát bez potřeby ruční úpravy
 * Výpočet těžiště, hmotnosti a dalších parametrů na základě GCODe. Následné uložení těchto parametrů do předchozích formátů (gcode komentář, AMF parametry, eSCAD metadata)
 * Tvorba celkových sestav, výpočet celkového těžiště, dalších parametrů

Celý systém by mělo být možné zasmyčkovat a pomocí nějakých variabilních parametrů a pomocí nějakých optimalizačních algoritmů moct přesně nastavit například přesné vyvážení dílu. Následně i celých sestav.

### Zdrojová data
Většina modelů je primárně v OpenSCADu. Ale existují i vyjímky, které by bylo dobré nějakým způsobem umět podporovat. 

Formát OpenSCAD navrhuji rozšířit o nějaké metatagy. (prozatím například použítím deklarátoru ```///@```). Vytvořit nějaký Extended OpenSCAD (```.escad```). Přes tyto metatagy by mělo jít nastavit všechny parametry potřebné k následným krokům tohoto postupu. 

> OpenSCAD nezná deklarátor ```@```. Proto ta kombinace s ```//``` což je OpenSCAD komentář. Pří použití zavináče OpenSCAD končí chybou. Pokud by tento postup byl vážně dobrý, tak věřím, že by se nám tento deklarátor povedlo prosadit přímo do OpenSCADu bez nutnosti použití komentářů.

Seznam potřebných funkcí/parametrů:

 * Pro renderování
   * ```@render()``` - vyrenderuje samotný soubor
   * ```@render(<module>)``` - vyrenderuje samotný modul
 * Pro nastavení slicování
   * ```@set_slicing_profile(<slicer.ini>)``` - Includování ini souboru pro konfiguraci sliceru. Lze importovat vícekrát. Vždy budou přepsané duplicitní hodnoty
   * ```@set_slicing_parameter(<parametr>, <hodnota>)``` - Nastavení jednoho parametru
 * Nastavení submodulu
   * ```set_modificator(<name>, <module>)``` - nastavit, který modul se použije jako modifikátor
 * Nastavení slicovacích parametrů pro slicer
   * ```set_modificator_profile(<modif_name>, <slicer.ini>)```
   * ```set_modificator_value(<modif_name>, <parametr>, <hodnota>)```

### Renderování modelů
Tento krok je věnován především řešení s OpenSCADem. Model pro tisk typicky obsahuje několik částí. Samotný model. Může být definován samotným souborem (výchozí stav) či  to může být konkrétní OpenSCAD module. Model také může obsahovat nějaké modifikátory. Což jsou taktéž 3D modely, vybírající část samotného modelu. 

Zatím by se OpenSCAD měl volat jako shell funkce, které bude předkládán OpenSCAD soubor. Popřípadě bude nějaký takový generován automaticky skriptem. 

Tyto kroky musí jít udělat i bez OpenSCAD dat. (například FreeCAD a jiných). U všech nelze zajistit, to, že půjde soubor tak jednoduše rozšířit. Proto by bylo dobré mít nějaký jednotný formát, které se přiloží např k FreeCADu a potřebná data se získají z něj.

### Skládání modelů
Z OpenSCADu vzniknou AMF modely, které je potřeba pro slicování sloučit do jednoho formátu. Vzhledem k tomu, že je to XML soubor. Stačí pouze jejich části překopírovat do jednoho souboru a nastavit správné parametry. Na základě deklarátorů v ```.escad``` souboru je pak vybráno, co bude modifikátor a co bude samotný model. Do výsledného AMF souboru pak budou přepsány požadované parametry z ```.ini``` souboru. A vznikne tak jeden kompaktní soubor pro slicování.

### Slicování 3D modelů
Zatím se pro slicování používá slicer [slic3r](). Tento slicer má několik vlastností, které jsou vvhodné pro toto použití.

Všechny parametry pro slicování by ideálně měly být uloženy v jednom souboru, který se Slic3eru předloží. Takovým souborem je formát ```.amf```. Což je formát založený na textovém ```.XML```. V souboru můžou být uloženy jak samotný model, jeho slicovací parametry tak i modifikátory s modifikujícími parametry.

Parametry i modifikátory budou uloženy přímo v tomto souboru. Což má vlastnost, že by tento model mělo být možné vyrenderovat bez závislosti na dalších souborech. A nebude potřeba řešit verze dalších souborů. 

Slic3er při procesu slicování dělá translace (kvůli vyrovnání na podložku). Tyto parametry jsou zatím neznámé a na získání těchto parametrů se pracuje v #279. 

### Výpočet těžiště a dalších parametrů reálného výrobku
Při vytištění je model vyplněn nějakým materiálem. Není to homogenní materiál a proto to nelze počítat z AMF či STL. Navíc každé slicování (s jiným slicerem) může být trošku odlišné. Tato věc by měla být dělána tak, že se budou počítat jednotlivé přejezdy extruderu a množství vytištěného materiálu. 

Z toho lze určitě získat těžiště a hmotnost modelu. Tyto parametry pak budou zapsány do komentáře gcodu a asi i do AMF souboru. 

### Sestavy

### Řešení
Celý problém by měl být řešený jako Python program/třída která bude provázet celým procesem. Jednotlivé kroky by měly být vlastní třídou. Ve výsledku by to měla být spustitelná aplikace.

Renderování pomocí makefilu bych zachoval. Jen by volal takovýto skript, který by tuto problematiku vyřešil. Makefile by se tím zásadně zjednodušil. Zároveň by to mělo být použitelné na jakýkoliv snad soubor bez nějakých dalších potřebných věcí... Něco jako ```escad_processor model.escad``` a mělo by vzniknout několik souborů. AMF s modifikátory, STL, png, Gcode. 

V průběhu by bylo dost důležité psát nějakou dokumentaci aby se takovýto software mohl snadno rozšířit. Věřím, že tento problém řeší mnoho lidí. Možná by následně bylo vhodné to někde prezentovat. 

Je zde možná nedořešeno dost detailů. Takže budu rád za připomínky, nápady.

kaklik, 2020-02-07:
Myslím že by bylo vhodné přeložit a alespoň části přidat sem: https://github.com/slic3r/Slic3r/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+openscad

roman-dvorak, 2020-02-09:
Jak takový SW nazveme? :)

kaklik, 2020-02-09:
> Jak takový SW nazveme? :)

No já myslím, že v tomhle případě bychom se spíše než na samotný vývoj takového nástroje měli soustředit na kvalitní zdokumentování toho, co potřebujem a  následný rozšíření tohoto nápadu do skupin, které by o to taky stály. Neboť na vývoj něčeho tak rozsáhlého momentálně nemáme kapacitu.