ThunderFly-aerospace/TF-G1 #284: Připevnění ventilu předních pístů

roman-dvorak, 2020-02-17:
Ještě bych Tě chtěl @cesilko1 poprosit o jednu úpravu středního podvozkového dílu. A to přimodelování držáčku na ventil od pístů. Důvod této úpravy je, aby tam ventil nějak moc nevlál popřípadě se nedřel o kolečko. 

Přibližné umístění je vidět na následujícím obrázku. Předpokladám,  že by se připevnil dvojicí M3 šroubů do otvoru, který by byl ve středu toho dílu. taková díra by měla zároveň celý díl vystužit. Matka by měla být zapuštěna z horní strany. Rozteč děr pro šrouby na ventilu je 16.5 mm. 

[image]

kaklik, 2020-03-02:
@roman-dvorak  je tohle stále aktuální?

roman-dvorak, 2020-03-02:
Tohle issue bych teď ukončil. S upraveným podvozkem a obráceně umístěnými písty by to připevnění mělo vypadat o dost jinak. Teď bych to však vyřešil sdrhovací páskou.

closed 2020-03-02 by roman-dvorak